openquantumhardware/qick #315: activate all the ADC ports in RFSoC4x2

zyw2362382, 2025-02-12:
Hi,
I am using qick package to control the FPGA RFSoC4x2. Although there are 4 ADC ports totally, only two ADC ports are activated in the QICK overlay. If I want to activate all the ADC ports, is there an existing bitstream that can support this? 
Thank you very much!

meeg, 2025-02-12:
Sorry, there is no existing bitstream - you would need to modify the firmware project, which is not trivial.

cpcepa, 2025-02-19:
Is there any plan to add in more ports for ADC in the qick_processor, e.g. enough ports to connect to all 16 ADC channels in ZCU216 (ZU49DR)

zyw2362382, 2025-02-19:
I  don't know if adding in more ports for ADC in the qick_processor works in this case, because I guess this configuration is defined by the bitstream, not defined by the QICK.

> Is there any plan to add in more ports for ADC in the qick_processor, e.g. enough ports to connect to all 16 ADC channels in ZCU216 (ZU49DR)

meeg, 2025-02-19:
There is nothing about the QICK firmware components that prevents using them to make a firmware with 16 ADCs. There is no plan to make such a firmware.

meeg, 2025-03-05:
Cleaning up old issues; feel free to reopen if you have some follow-up.

closed 2025-03-05 by meeg