openquantumhardware/qick #319: Only I value readout on freshly built tprocv2 4x2 project.

seangal2, 2025-03-02:
Hi,

I am working on Qick for a few months now, and I just noticed that when I preform readout using my custom firmware I only get I values (Q=0). 
I then rebuilt a fresh new project based on the latest files for the tprocv2 4x2 project, and saw the same behavior.

Looking at the code, the readout block that performs demodulation and down conversion should produce IQ outputs.

Testing the readout on the packaged firmware (which is tprocv1) I get both I and Q.

Is this a known issue?

meeg, 2025-03-02:
This is not expected.

If you use https://github.com/meeg/qick_demos_sho/blob/main/tprocv2/00_intro.ipynb (both the Python code and the 4x2 firmware linked from the notebook), does it work?

seangal2, 2025-03-02:
Hi,

Thanks for the link.
I used the notebook (and firmware) and measured the IQs correctly. 
By looking at the code, I realized that I was missing the `send_readoutconfig` instruction in my program (only had `add_readoutconfig`).

fixed that and now I ran my IQ test program (which sends a grid of IQ values and reads them) and got the following result:
[image]

For V1 I got:

[image]

Looks like in V2 there is a phase drift between the generator and the readout. Increasing the `number_samples` per IQ from 10 to 100+ produces full circles. On V1 this was not observed. 

Additionally, when fine-tunning the `adc_trig_offset` I saw that the timing (not just the phase) is inconsistent, which I guess comes from the same root cause.

This is the V2 program for reference:
```
class LoopbackProgram(AveragerProgramV2):
    def _initialize(self, cfg):
        res_ch = cfg["res_ch"]

        # set the nyquist zone
        if cfg.get('has_mixer', False):
            self.declare_gen(ch=cfg["res_ch"], nqz=1, mixer_freq=cfg['mixer_freq'], ro_ch=cfg["ro_chs"][0])
        else:
            self.declare_gen(ch=cfg["res_ch"], nqz=1, ro_ch=cfg["ro_chs"][0])
        
        # configure the readout lengths and downconversion frequencies (ensuring it is an available DAC frequency)
        for ch in cfg["ro_chs"]:
            self.declare_readout(ch=ch, length=self.cfg["readout_length_us"])

        self.add_readoutconfig(
            ch=cfg["ro_chs"][0],
            name="readout",
            freq=cfg["pulse_freq"],
            phase=0
        )
        self.send_readoutconfig(
            ch=cfg["ro_chs"][0],
            name="readout"
        )
            

        self.iq_size = cfg['iq_size']

        i = 0
        for I in np.linspace(-1, 1, self.iq_size):
            for Q in np.linspace(-1, 1, self.iq_size):
                gain = np.sqrt(I ** 2 + Q ** 2) / np.sqrt(2)
                gain = gain * self.cfg['pulse_gain']
                phase = np.arctan2(Q,I) * 180 / np.pi
                self.add_pulse(
                    ch=res_ch,
                    name="pulse_"+str(i),
                    freq=cfg["pulse_freq"],
                    phase=phase,
                    style='const',
                    gain=gain,
                    length=cfg['readout_length_us']
                )
                i += 1

    def _body(self, cfg):
        res_ch = cfg["res_ch"]
        for i in range(self.iq_size**2):
            self.pulse(ch=res_ch, name='pulse_'+str(i))
            self.trigger(
                ros=cfg['ro_chs'],
                pins=[0],
                width=cfg['readout_length_us'],
                ddr4=True,
                mr=True,
                t=cfg['adc_trig_offset']
            )
            self.delay_auto()
            self.delay(1.0)
        self.wait_auto()

GEN_CH = 1
PUMP_CH = 0
RO_CH = 0
FREQ = 2000 # MHz
N_TRANSFERS = 10000

soc.reset_gens()
config={"res_ch":GEN_CH, # --Fixed
        "ro_chs":[RO_CH], # --Fixed
        "reps":1, # --Fixed
        "relax_delay":1.0, # --us
        
        "length":30000, # [Clock ticks]
        # Try varying length from 10-100 clock ticks
        
        "readout_length":10, # [Clock ticks]
        "readout_length_us": soc.cycles2us(1),
        # Try varying readout_length from 50-1000 clock ticks

        "pulse_gain": 0.1, # [-1 - 1]
        # Try varying pulse_gain from 500 to 30000 DAC units

        "pulse_freq": FREQ, # [MHz]
        # In this program the signal is up and downconverted digitally so you won't see any frequency
        # components in the I/Q traces below. But since the signal gain depends on frequency, 
        # if you lower pulse_freq you will see an increased gain.

        "adc_trig_offset": soc.cycles2us(180), # [Clock ticks]
        # Try varying adc_trig_offset from 100 to 220 clock ticks

        "soft_avgs":1,
        "mixer_freq": 2900,
        "has_mixer": soccfg['gens'][0]['has_mixer'],

        "iq_size": 3,
       }

number_samples = 10

readout_length = 100

assert readout_length < config['length']
print(readout_length)
config['readout_length'] = readout_length
config['readout_length_us'] = soc.cycles2us(readout_length)
prog = LoopbackProgram(soccfg, reps=config['reps'], cfg=config, final_delay=config['relax_delay'])
prog.config_all(soc)

# Arm the buffers
soc.arm_ddr4(ch=RO_CH, nt=N_TRANSFERS)
soc.arm_mr(ch=RO_CH)

# Run the program
iq_lists = []
for i in range(number_samples):
    iq_list = prog.acquire(soc, soft_avgs=config['soft_avgs'], progress=False)
    iq_lists.append(iq_list)
iq_lists = np.stack(iq_lists)
print(iq_lists.shape)

# Analyze
avg_IQ_points = iq_lists[:,0,:, 0] + 1j * iq_lists[:,0,:,1]

avg_avg_IQ_points = np.average(avg_IQ_points, axis=0)
std_avg_IQ_points = np.sqrt(np.std(np.real(avg_IQ_points), axis=0) ** 2 + np.std(np.imag(avg_IQ_points), axis=0) ** 2)

indxes = list(range(avg_IQ_points.shape[1]))
random.shuffle(indxes)

for i in indxes:
    plt.scatter(np.real(avg_IQ_points[:,i]), np.imag(avg_IQ_points[:,i]))

ax = plt.gca()
for avg, std in zip(avg_avg_IQ_points, std_avg_IQ_points):
    ax.add_patch(plt.Circle((np.real(avg), np.imag(avg)), std, color='r', fill=False))
plt.show()
```

meeg, 2025-03-02:
Phase drift - your add_pulse and add_readoutconfig are missing the frequency-matching parameters. I suggest that you should check this behavior with minimal changes from the example code, before writing your own code.

Timing - need more detail, I think this is a separate issue.

seangal2, 2025-03-02:
Phase drift - oh you are right, I missed that in the migration from V1 to V2 (and I didn't see your V2 examples so I tried figuring this from source) 

The results now match my V1 measurement (below is V2):
[image]

Timing - 
When I run the code below (with the same program), for some runs I get see the rising edge, for some the falling edge. Note that this is jitter in the order of a clock cycle. (of course I can time the measurement to not include the edges, but this shows that there is some jitter between the channels)
[image]

```
soc.reset_gens()
config={"res_ch":GEN_CH, # --Fixed
        "ro_chs":[RO_CH], # --Fixed
        "reps":1, # --Fixed
        "relax_delay":1.0, # --us
        
        "length":30000, # [Clock ticks]
        # Try varying length from 10-100 clock ticks
        
        "readout_length":50, # [Clock ticks]
        "readout_length_us": soc.cycles2us(50),
        # Try varying readout_length from 50-1000 clock ticks

        "pulse_gain": 0.1, # [-1 - 1]
        # Try varying pulse_gain from 500 to 30000 DAC units

        "pulse_freq": FREQ, # [MHz]
        # In this program the signal is up and downconverted digitally so you won't see any frequency
        # components in the I/Q traces below. But since the signal gain depends on frequency, 
        # if you lower pulse_freq you will see an increased gain.

        "adc_trig_offset": soc.cycles2us(180), # [Clock ticks]
        # Try varying adc_trig_offset from 100 to 220 clock ticks

        "soft_avgs":1,
        # Try varying soft_avgs from 1 to 200 averages
        "pump_ch": PUMP_CH,
        "pump_freq": FREQ-10,
        "pump_gain": 1,
        "pump_on": False,

        "mixer_freq": 2900,
        "has_mixer": soccfg['gens'][0]['has_mixer'],

        "iq_size": 3,
       }

prog = LoopbackProgram(soccfg, reps=config['reps'], cfg=config, final_delay=config['relax_delay'])
prog.config_all(soc)
N_TRANSFERS2 = 10
# Arm the buffers
soc.arm_ddr4(ch=RO_CH, nt=N_TRANSFERS2)
soc.arm_mr(ch=RO_CH)

# Run the program
for i in range(10):
    iq_list = prog.acquire_decimated(soc, soft_avgs=config['soft_avgs'], progress=False)
    # iq_list = prog.acquire_decimated(soc, soft_avgs=10)
    t = prog.get_time_axis(ro_index=0)
    
    # plt.plot(t, iq_list[0][0,:,0], label="I value")
    # plt.plot(t, iq_list[0][0,:,1], label="Q value")
    plt.plot(t, np.abs(iq_list[0][0,:,:].dot([1,1j])), label=f"magnitude {i}")
    plt.legend()
    plt.ylabel("a.u.")
    plt.xlabel("us");
```

meeg, 2025-03-03:
That's expected when the readout's decimated data clock is different from the tProcessor timing clock - the readout trigger goes through a clock domain crossing from the tProc to the readout, and you get jitter there.

closed 2025-03-03 by seangal2