openquantumhardware/qick #326: Minimum/Maximum gain value warnings

ChunyangDing, 2025-04-15:
On the [Firmware documentation page](https://qick-docs.readthedocs.io/latest/firmware.html) the maximum gain range for any particular DDS is listed as 16-bit signed ([-32768,32767]). This should still be true for current QICK firmwares, but as of asm_v2, gain is now given as a simple percentage between -1 and 1, where the value is automatically scaled. 

An issue can occur if operating below the minimum resolution of the gain, < 3e-5, where the user believes that they are generating a small pulse, but the rfsoc actually is rounding it to zero. No warning is given at this point in the code, and it's also a bit tricky to find what the actual maxv value is. (I found it by soccfg['gens'][gench=0]['maxv'], but this took a bit of digging - it's not output as part of the standard boilerplate of soccfg, for example, and is especially confusing given that the boilerplate says "32-bit dds" without clarification that it's in frequency). 

Likewise, no error is given if gain > 1, even though it certainly gets rounded down to just be 1. This might be confusing for new users who are not aware of the limitations of gain.

meeg, 2025-04-15:
Yes, already aware - but these are good points and thanks for writing them down.

rylanmalarchick, 2026-10-16:
One thing to add on gain > 1: on asm_v2 it does not get rounded down to 1. The value goes into the register as is, past `maxv`.

qick 0.2.436, ZCU216 config, compile only:

| gen | requested gain | register value | maxv |
|-----|----------------|----------------|------|
| `axis_signal_gen_v6`, const pulse | 1.5 | 49149 | 32766 |
| `axis_sg_mixmux8_v1`, `mux_gains=[1.5]` | 1.5 | 49149 | 32766 |

49149 is outside the signed 16-bit gain range from the firmware docs. A `RuntimeError` for |gain| > 1 in `add_pulse` and `declare_gen` would catch it at compile time.

closed 2026-11-06 by meeg