openquantumhardware/qick #346: Synthesis error in `axis_signal_gen_v6` while building `qick_tprocv1_216_standard`

ashuping, 2025-06-27:
Hello! I've been trying to get a reference project built for QICK. I was able to run the `bd_2022-1.tcl` script to create the project, but it seems to be failing during synthesis.

## Steps to Reproduce
1. Clone QICK (`main` branch - at the time of writing, that's commit `4770f84`)
2. Open Vivado 2022.1
3. Run the `qick_tprocv1_216_standard` script. It creates the project, but it exits part-way through because some of the project dependencies are missing
```
cd <qick-clone-dir>/firmware/projects/qick_tprocv1_216_standard
source ./bd_2022-1.tcl
```
4. Add the contents of `firmware/hdl` to the project sources
5. Add `firmware/ip` to the project IP repositories
6. Re-run the script (`source ./bd_2022-1.tcl`). It succeeds this time
7. Generate an HDL wrapper for the generated design
8. Set the generated wrapper as top-level
9. Try to run synthesis/implementation

## Expected Result
The project synthesizes and implements successfully.

## Actual Result
Seven instances of `axis_signal_gen_v6` encounter errors as Vivado attempts to launch synthesis runs:
```
ERROR: [IP_Flow 19-8166] Failed to verify json document against the schema 'xilinx.com:schema:json_instance:1.0'.
{
  "additionalProperties": {
    "disallowed": "cell_name",
    "instanceRef": "#/ip_inst",
    "schemaRef": "file:///json_instance_v1_0.json#/properties/ip_inst"
  }
}
CRITICAL WARNING: [IP_Flow 19-979] Failed to recreate IP instance 'dds_compiler_0'. Error setting original project options.
ERROR: [IP_Flow 19-3505] IP Generation error: Failed to generate IP 'axis_signal_gen_v6_0'. Failed to create IP subcores.
ERROR: [IP_Flow 19-98] Generation of the IP CORE failed.
Failed to generate IP 'axis_signal_gen_v6_0'. Failed to create IP subcores.
```

The full log file can be found here: [vivado_qick_synth_failure_annotated.log](https://github.com/user-attachments/files/20955623/vivado_qick_synth_failure_annotated.log) (Note that I removed the comments that are just echoing the lines of the script for readability purposes. I also replaced all instances of my actual project directory with `$PROJECT_DIRECTORY`).

meeg, 2025-06-27:
You should not be running the `bd_2022-1.tcl` script directly; you should be running the `proj.tcl` script following the instructions in https://github.com/openquantumhardware/qick/blob/main/firmware/README.md.

ashuping, 2025-06-27:
Looks like that fixed it. Thanks for the help!