openquantumhardware/qick #361: repeated value in axis_avg_buffer data

meeg, 2025-08-21:
#337 seems to have introduced a bug where one value is repeated in the data pulled from the accumulated and decimated buffers.

I am testing with these ZCU111 firmwares:
* https://s3df.slac.stanford.edu/people/meeg/qick/tprocv2/2025-05-30_111_tprocv2r23_standard/ - axis_avg_buffer v1.1
* https://s3df.slac.stanford.edu/people/meeg/qick/tprocv2/2025-06-26_111_tprocv2r25_standard/ - axis_avg_buffer v1.2

Test program, plays a triangle-envelope pulse where out-of-order samples will be visible:
```
class RampProgram(AveragerProgramV2):
    def _initialize(self, cfg):
        ro_ch = cfg['ro_ch']
        gen_ch = cfg['gen_ch']
        self.declare_gen(ch=gen_ch, nqz=1)
            
        self.add_triangle(gen_ch, "myenv", cfg['pulse_len'])
        self.add_pulse(ch=gen_ch,
                      name="mypulse",
                      ro_ch=ro_ch,
                      style="arb",
                      freq=cfg['freq'],
                      envelope="myenv",
                      phase=cfg['phase'],
                      gain=cfg['gain'],
                      )

        self.declare_readout(ch=ro_ch, length=cfg['ro_len'])

        self.add_readoutconfig(ch=ro_ch,
                              name="myro",
                              freq=cfg['freq'],
                              gen_ch=gen_ch)

        self.send_readoutconfig(ch=ro_ch,
                                   name="myro",
                                   t=0)

    def _body(self, cfg):
        self.delay(1.0)
        self.pulse(ch=cfg['gen_ch'], name="mypulse", t=0)
        self.trigger(ros=[cfg['ro_ch']], pins=[0], t=cfg['trig_time'], ddr4=True)
        
config = \
{
    'gen_ch': 7,
    'ro_ch': 0,
    'freq': 500,
    'trig_time': 0.35,
    'ro_len': 0.1,
    'pulse_len': 0.1,
    'gain': 1.0,
    'phase': 0.0
}
```

## Test the decimated buffer
```
prog = RampProgram(soccfg, reps=1, final_delay=0.5, cfg=config)

iq_list = prog.acquire_decimated(soc)
plt.plot(iq_list[0][:,0], label='I')
plt.plot(iq_list[0][:,1], label='Q')
plt.xlabel('time [samples]')
plt.ylabel('amplitude [arb. units]')
plt.legend();
```

v1.1, looks OK:
[image]

v1.2, we don't get sample 8, instead we see sample 9 twice:
[image]

## Test the accumulated buffer
```
prog = RampProgram(soccfg, reps=20, final_delay=0.5, cfg=config)
iq_list = prog.acquire(soc)
print(prog.get_raw())
```

v1.1, looks OK:
```
[array([[[ 6471, 12456]],
       [[ 6416, 12383]],
       [[ 6374, 12274]],
       [[ 6463, 12466]],
       [[ 6412, 12372]],
       [[ 6356, 12277]],
       [[ 6470, 12471]],
       [[ 6407, 12374]],
       [[ 6364, 12270]],
       [[ 6481, 12457]],
       [[ 6402, 12382]],
       [[ 6341, 12291]],
       [[ 6453, 12473]],
       [[ 6418, 12365]],
       [[ 6380, 12268]],
       [[ 6453, 12476]],
       [[ 6411, 12387]],
       [[ 6348, 12299]],
       [[ 6463, 12457]],
       [[ 6402, 12379]]])]
```

v1.2, [4908, 13100] is repeated:
```
[array([[[ 4902, 13081]],
       [[ 4895, 13014]],
       [[ 4845, 12920]],
       [[ 4929, 13096]],
       [[ 4886, 12997]],
       [[ 4830, 12906]],
       [[ 4894, 13107]],
       [[ 4885, 13008]],
       [[ 4908, 13100]],
       [[ 4908, 13100]],
       [[ 4886, 13015]],
       [[ 4860, 12910]],
       [[ 4912, 13082]],
       [[ 4873, 13011]],
       [[ 4825, 12920]],
       [[ 4917, 13097]],
       [[ 4884, 13030]],
       [[ 4842, 12947]],
       [[ 4905, 13112]],
       [[ 4886, 13019]]])]
```

mmdiego, 2025-08-21:
I'll check it

mmdiego, 2025-08-21:
I was able to replicate the bug in hardware. Everything seems to point to a problem in the handshaking between the DMA and the buffers read when the READY toggles. However, I don't see how that could be related with the fix I did, as #337 changed something in the write process and not in the read sequence.

[image]

[image]

I also ran a simulation and keeping tready high I'm not able to reproduce the bug, which makes sense to me. Randomly toggling tready seems to corrupt the output sequence, so I'll fix that behavior and we can give it a try again.

@meeg is there any other considerable difference between the 2 firmwares you are comparing? maybe something in the clock frequencies that may lead to a different behavior in how the data is being transferred to the dma?

mmdiego, 2025-08-22:
Waveform from simulation `test_issue361` replicating the bug

[image]

mmdiego, 2025-08-22:
Waveform from same simulation after fix applied:

[image]

closed 2025-08-22 by meeg